hifiberry/hifiberry-os #554: Pause Shairport-Sync from Beocreate UI

LukeWinikates, 2024-06-01:
**Is your feature request related to a problem? Please describe.**

I primarily use my Hifiberry speakers as Airplay 2 devices via Shairport Sync. If I start playing music from an iOS device, I can't pause it from the web UI. If I press the pause button, the icon temporarily turns into a 'play' icon, then reverts to a pause icon within a few seconds. The audio never stops playing.

hifiberryos version: 20240319

**Describe the solution you'd like**

I would like pausing in the UI to allow pausing the the Airplay2 audio stream.

**Describe alternatives you've considered**

I can restart the Hifiberry device, which does cause the audio to stop. Or I can find the iOS device and pause it, but that's inconvenient.

**Additional context**

I read through the source code in `bang-olufsen/create`. I can see that the play/pause sends a websocket message to the `node` server. Via `journalctl`, I can see that the `node` app sends a message to the controller:

```
Jun 01 18:11:33 hifi-office python3[409]: INFO: controller - sent Pause to org.mpris.MediaPlayer2.ShairportSync
Jun 01 18:11:33 hifi-office python3[409]: 127.0.0.1 - - [01/Jun/2024 18:11:33] "POST /api/player/playpause HTTP/1.1" 200 2
```

Play/pause works if I use the `radio` source:

```
Jun 01 18:27:14 hifi-office python3[409]: INFO: controller - sent Pause to mpd
Jun 01 18:27:14 hifi-office python3[409]: 127.0.0.1 - - [01/Jun/2024 18:27:14] "POST /api/player/playpause HTTP/1.1" 200 2
Jun 01 18:27:15 hifi-office node[748]: Source 'radio' has deactivated.
```

The UI seems to be working properly, but either the python3 `controller` is not sending the right message to `mpris` clients or to `org.mpris.MediaPlayer2.ShairportSync` specifically, or the `mpris` implementation in ShairportSync is not handling pause messages correctly.
I'm happy to debug further. I'll have to dig deeper into `mpris` to do that effectively. Curious if the `hifiberry` maintainers have thoughts about this issue.

LukeWinikates, 2024-06-01:
I think I found the root cause of this in Shairport-sync mikebrady/shairport-sync/issues/1822

It's tied to changes in the remote control protocol for airplay in newer versions of iOS/macOS. I have an older mac running Monterey and play/pause works fine there.

hifiberry, 2024-06-04:
Unfortunately in this case, we can't do anything about it. When this gets reverse engineered again and integrated into shairport, it will work again in HBOS, but we don't know if and when this will happen.

closed 2024-06-04 by hifiberry