hifiberry/hifiberry-os #556: Volume control with number showing

Raitsa, 2024-07-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Volume control with number showing.

It would be very user friendly if the volume control would have numbers ranging from 0 up to 100.
And perhaps an option to set the maximum volume to any value from 0 to 100 to make sure no speakers are destroyed accidentally.
The current slider to control volume output is somewhat of a hassle to use.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hifiberry, 2024-07-05:
Hmm, I'm not sure what a number would help. Just think of 

left = 0
center = 50
right = 100

How would showing this number help you?

Raitsa, 2024-07-05:
I find the included example very intuitive and reminds me of an analog volume knob we all know, the example is the volume control in rAudio.
I have multiple Raspberries with Hifiberry64 and they have different (active) speakers attached to them. Some are set to volume 15, some are set to more than that. So they behave differently regarding the default volume amount. A number is way easier to set and remember, atleast to my brain :-)

Also adding volume just the least amount, meaning +1 is way easier with a plus and minus than with a slider on a touchscreen or on a browser.

+ to add volume.
- to decrese volume.
- middle button to mute

[image]

hifiberry, 2024-07-29:
We'll have a look, but UI changes like this are complex as they need to work on different devices and resolutions.

We won't integrate a numerical value as there is no good "meaning" for a number. 15 on one device can be completely different than 15 on another. Therefore, we think this would create even more ambiguity.

If the output volume of HiFIBerryOS is way to high at full volume, I recommend to limit the volume:
https://www.hifiberry.com/blog/limiting-the-maximum-output-volume-in-hifiberryos/

Also note that while you can set the volume in the UI, most services can override this setting and change the volume independent from the UI.