hifiberry/hifiberry-os #560: Raspberry Pi 5 w/ Official 7" Touchscreen Support

Cbeck527, 2024-07-28:
**Describe the bug**
There are display issues using the following hardware:
* Raspberry Pi 5
* Official 7" Touchscreen Support

First, the display does not function until overlay is added. Then, the now playing display appears to be "zoomed in" too much, as if it does not fit the screen.

**HiFiBerryOS version**
64alpha8
https://github.com/hifiberry/hifiberry-os/releases/tag/64alpha8
`hifiberryos-20240523-pi4.img`

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Install HifiBerry on Raspberry Pi 5 with official 7" touchscreen
1. Display does not function
1. Add

**Expected behavior**
* Functioning display
* Web browser fitting content appropriately.

**FIX**
I was able to fix both issues with the following steps:

1. Remount filesystem as read/write:
```shell
mount -o remount,rw /boot
```
2. Add line under to [pi5] section
```conf
[pi5]
dtoverlay=vc4-kms-v3d-pi5,noaudio
# CUSTOM DISPLAY SETTINGS
dtoverlay=vc4-kms-dsi-7inch,dsi0,sizex=800,sizey=480
# END CUSTOM DISPLAY SETTINGS
```
3. Create system unit override to set cog environment variable for wayland (note: I had to use `nano` as the editor... there seems to be an issue with the built-in vim)
```
EDITOR=nano systemctl edit cog
```
4. Add the following to the unit override
```systemd
[Service]
Environment=COG_PLATFORM_WL_VIEW_FULLSCREEN=1
```
5. Run `systemctl restart cog` or reboot

hifiberry, 2024-07-29:
Thank you for the contribution. We'll see how we can integrate this into the next release

Hans-Haefner, 2024-08-18:
System unit override to set cog environment variable for wayland also works for my Luckfox 7" HDMI LCD Display (1024x600)

user10072023github, 2024-09-12:
Can someone walk me through how to do this when putting the SD card with HiFiBerryOS installed on it, into my Ubuntu box?  Using sudo nautilus to avoid permissions issues but can't find these files to edit...

Profoktor, 2024-12-19:
I'm having the same issue with a RPi 3B+. I tried the steps as described here (adjusting the config.txt to load the overlay for any pi). 

This results in the display being recognized, but I only get a login terminal, and when restarting the cog-service I get an error that a required service failed:

> journalctl -xe
> Dec 19 12:53:57 musicstation systemd[1]: Starting Weston, a Wayland compositor, as a system service...
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.957 [INFO] [seatd/server.c:145] New client connected (pid: 2338, uid: 0, gid: 0)
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.957 [INFO] [seatd/seat.c:170] Added client 1 to seat0
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.957 [INFO] [seatd/seat.c:480] Opened client 1 on seat0
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.959 [INFO] [seatd/seat.c:418] No clients on seat0 to activate
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.963 [INFO] [seatd/seat.c:524] Closed client 1 on seat0
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.963 [INFO] [seatd/seat.c:192] Removed client 1 from seat0
> Dec 19 12:53:57 musicstation seatd[270]: 00:08:10.963 [INFO] [seatd/client.c:471] Client disconnected
> Dec 19 12:53:57 musicstation systemd[1]: weston.service: Main process exited, code=exited, status=1/FAILURE
> Dec 19 12:53:57 musicstation systemd[1]: weston.service: Failed with result 'exit-code'.
> Dec 19 12:53:57 musicstation systemd[1]: Failed to start Weston, a Wayland compositor, as a system service.
> Dec 19 12:53:57 musicstation systemd[1]: Dependency failed for Local web brower to display web interface.
> Dec 19 12:53:57 musicstation systemd[1]: cog.service: Job cog.service/start failed with result 'dependency'.q
>

Profoktor, 2024-12-21:
@Cbeck527 - did you notice if the screensaver is working in your setup?

It's not working for me, but after changing the overlay it might also be related to this...