hifiberry/hifiberry-os #565: hifiberryos-alpha8.zip attached display not working

pdavies91, 2024-08-26:
**Describe the bug**
Raspberry pi 3b+
Running originally the 20230404, works fine (with some niggles). 
Installed hifiberryos-alpha8.zip onto another card inserted and started.
Rainbow screen comes and goes.... then rotates through white, green, red .....screens 

**HiFiBerryOS version**
hifiberryos-alpha8.zip

**HiFiBerry sound card**
DAC+/Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. INstall SD card and power up

**Expected behavior**
Should display HifiberrOS interface

user10072023github, 2024-09-14:
Seems to be a common theme...I'd stay on the old version if that's working ok for ya.  Wish the team would get back to folks.

hifiberry, 2024-09-19:
See https://github.com/hifiberry/hifiberry-os/issues/560

closed 2024-09-19 by hifiberry

pdavies91, 2024-10-01:
Moved to Moode Audio