hifiberry/hifiberry-os #566: Apple Superdrive support for playing CDs

fiftysomethingguy, 2024-09-03:
**Is your feature request related to a problem? Please describe.**
USB Apple Superdrive is not supported for playing CDs

**Describe the solution you'd like**
USB Apple superdrive support to be added

**Describe alternatives you've considered**
Using another CD drive but they're not as cool

**Additional context**
@bweston6 The USB slot loading Apple Superdrive requires a bytestring sent to it before it will wake up and do anything - it won't even suck in a disc without it. Macs do this automatically. It would require the sg3_utils package to be installed and an sg_raw write command issued. This can be done by adding a rule to udev that detects the manufacturer and device ID when it is plugged in.

ACTION=="add", ATTRS{idProduct}=="1500", ATTRS{idVendor}=="05ac", DRIVERS=="usb", RUN+="/usr/bin/sg_raw %r/sr%n EA 00 00 00 00 00 01"

The Superdrive also has no eject button. CDs are ejected on Mac by dragging them to the Trash can. The HiFiBerry UI would need an eject button.

bweston6, 2024-09-03:
Thanks! I'll have a look at this at the end of the week

bweston6, 2024-09-11:
@hifiberry looks like this would need https://github.com/hifiberry/create/pull/16 to be merged if you could take a look please. I think it would be best to put the eject button on that source page, unless you'd want it on the now playing screen somewhere.