hifiberry/hifiberry-os #569: Rotar

discoluc, 2024-09-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

borkog, 2024-11-30:
Related error:

ModuleNotFoundError: No module named 'pigpio'

borkog, 2024-11-30:
this means that "controller" is not working also because "pigpio" is not available