hifiberry/hifiberry-os #571: DAC+ADC board not recognized automatically upon installation of the OS (20230404) on Pi4

Tostibrand, 2024-09-11:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
DAC+ADC & Aamp60

**To Reproduce**
Steps to reproduce the behavior:
1. flash latest (20230404) OS to SDcard
2. Boot
3. Go to about
4. recognized as Dac+/Aamp2

**Expected behavior**
Recognized as Dac+Adc/Aamp60

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
I have tried to edit the config.txt after removing SDcard from pi4. to edit dtoverlay=hifiberry-dacplus to --> dtoverlay=hifiberry-daplusadc as suggested in earliers thread. But the change is not consistent as when i reinsert the SDcard and boot the config.txt is changed back to dacplus for dacplusadc. ive also tried to change the DT overlay on a fresh install and that one doesnt even boot at all

hifiberry, 2024-09-14:
Unfortunately we can't fix this as the AC can't be detected in software, as only the DAC on the DAC+ADC offers a software interface.  While it is annoying, we can't really fix it.

closed 2024-09-14 by hifiberry

Tostibrand, 2024-09-14:
I understand, I found a work around for this. Took a while to piece together so I'll post the fix here for other people to find in one place.

HOW TO FIX

It's really simple. 

SSH to you hifiberry and login

**SSH root@hifiberry.local**

Or

**SSH root@[ip4 adress of your hifiberry]** for example **SSH root@192.168.88.176**

Password: hifiberry

You then need to use this mount command otherwise the config.txt file we need to edit is read only. I've tried to edit the file on PC but then it doesn't boot or the change is not persistent.

**mount -o remount,rw /boot**

After that you type this to edit config.txt

**Nano /boot/config.txt**

Go down to the very bottom of the file and append **adc** to **dtoverlay=hifiberry-dacplus** so it becomes **dtoverlay=hifiberry-dacplusadc**

Press ctrl + X

It will ask to save, press **y** and hit enter to overwrite the file. Reboot (I used restart from web UI, maybe u can use **reboot** in terminal. Haven't tried) and Bob's your uncle

You should now have the analogue input source available in the UI under "disabled sources" from there just enable it and enjoy!