hifiberry/hifiberry-os #572: Overlay won't expand to fullscreen when loading into HiFiBerryOS

user10072023github, 2024-09-12:
**Describe the bug**
Tried 2 different 1920x1080 monitors and although the resolution seems to be detected fine, by the time the OS loads fully only 1/3 is the screen in the upper-right shows content.  All this started happening with the 64 alphas.  I've tried 1, 5, and 8.

**HiFiBerryOS version**
20240523

**HiFiBerry sound card**
DAC-Plus

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest OS on RPi4 with 1920x1080p monitor connected
2. Load into OS environment
3. Content not expanded to take up the full screen space even though cursor can go outside the content, indicating an overlay issue

**Expected behavior**
Content fills display's available space

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
Don;t know if this is helpful in the least, but some folks were saying here: https://forums.raspberrypi.com/viewtopic.php?t=305690 that 64-bit didn't support firmware kms yet...is that the issue happening here?  Thanks

user10072023github, 2024-09-12:
P.S. the reason I need to get these alphas to work is because I just bought the DAC-Plus and DSP board, installed it...only get little tiny mouse volume when on 20230404 and display works fine.  I install 20240523 and sound works like it should at good volume.  Ok, gotta get over to the alphas to get the volume I need but display issue...with old OS versions I get mouse sound (just the faintest sound with full volume) and no display issue.  Jeez, really need one of these issues fixed so I can just use whichever one works.

If anyone knows why I get mouse squeaks out of 20230404 with my brand new DAC-Plus and DSP on RPi4 please let me know.  Again, not hardware issue as I thought initially because putting 20240523 gets me normal sound out of my headphones.  Thanks

user10072023github, 2024-09-14:
Is there really no one else that encountered this issue?  This happens on my LG 1920x1080 desktop monitor, my Waveshare portable 1920x1080 display, and my Waveshare portable 1280x720 display.  Mouse can travel across full space of all displays when loaded into OS environment, but yet the OS content only shows in about 1/3 of the screen space.

user10072023github, 2024-09-14:
Gonna write this hifiberry team off.  They jump in just to mark a legit issue (updating their inaccurate/incomplete documentation or their site) as closed with insane response and have no reply for this like "aware of issue", "awaiting new kernel driver from 3rd party", etc.  Ridiculous

hifiberry, 2024-09-19:
See https://github.com/hifiberry/hifiberry-os/issues/560

closed 2024-09-19 by hifiberry