hifiberry/hifiberry-os #576: Spotify Connect fails

jakubstrama, 2024-09-12:
**Describe the bug**
Spotify Connect does not work anymore

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
DAC2 HD

**To Reproduce**
Steps to reproduce the behaviour:
1. Open any Spotify client
2. try to connect to Hifiberry via Spotify Connect
3. Status displays connecting
4. Log in via ssh to linux.
5. run "systemctl status spotify.service"
6. It displays : hifiberry vollibrespot[1248]: [Vollibrespot] : Connecting to AP "ap-gew4.spotify.com:443" 

**Expected behaviour**
It should be possible to connect via Spotify Connect

**Browser (if applicable)**
 - MacOs Spotify app 
 - iOS Spotify app

**Additional context**
This issue on librespot indicates, that Spotify does not support authentication via username/password anymore. Maybe it is somehow related.
[librespot/issues/1308](https://github.com/librespot-org/librespot/issues/1308)

rum0s, 2024-09-12:
I have the same issue. Switched to spotifyd for now.

jakubstrama, 2024-09-12:
> spotifyd

Do you run it alongside hifiberry-os on RaspberryPi? If so, would you share any details?

rum0s, 2024-09-12:
> > spotifyd
> 
> Do you run it alongside hifiberry-os on RaspberryPi? If so, would you share any details?

Yes. I disabled spotify connect as described in the blog and activated spotifyd via extensions. Afterwards I entered my credentials.

gjvg56, 2024-09-13:
I also have this problem with SpotifyConnect. For me it is under an older version of HifiberryOS . I'm afraid to change to HifiBerry64 where I could use SpotifyD. Or should I try this ?

rum0s, 2024-09-13:
> I also have this problem with SpotifyConnect. For me it is under an older version of HifiberryOS . I'm afraid to change to HifiBerry64 where I could use SpotifyD. Or should I try this ?

This is what I did. I upgraded to HifiberryOS64 from a older version from 2023. Just use a second SD card and give it a try.

gjvg56, 2024-09-13:
> 
> This is what I did. I upgraded to HifiberryOS64 from a older version from 2023. Just use a second SD card and give it a try.

Works like a charm. At first it didnt work, cos in HifyberryOS64 I logged in with spotify-credentials..not needed for the Deamon !!

diederikdq, 2024-09-13:
Hi
This is discussed here https://support.hifiberry.com/hc/en-us/community/posts/21348779832605-Spotify-Connect-doesn-t-work-anymore-HifiberryOS-20230404?page=1. Chances are your issue will most likely have to do with the /etc/hosts file where ap-gew4.spotify.com is set to resolve to 104.199.65.124 . But that IP won't work.

So either comment out the line: 
`nano /etc/hosts` and change `104.199.65.124 ap-gew4.spotify.com` into `#104.199.65.124 ap-gew4.spotify.com`
OR
fix it via the code where it got set.
`nano /usr/lib/systemd/system/spotify.service` and remove the line `ExecStartPre=/opt/hifiberry/bin/set-host-ip ap-gew4.spotify.com 104.199.65.124`
then remove the current config `nano /etc/hosts` and remove line `104.199.65.124 ap-gew4.spotify.com`
then reboot.

Repo wise, th is solved here: https://github.com/hifiberry/hifiberry-os/pull/577 but it will be for the 64 version I guess.

maxbundscherer, 2024-09-15:
> Hi This is discussed here https://support.hifiberry.com/hc/en-us/community/posts/21348779832605-Spotify-Connect-doesn-t-work-anymore-HifiberryOS-20230404?page=1. Chances are your issue will most likely have to do with the /etc/hosts file where ap-gew4.spotify.com is set to resolve to 104.199.65.124 . But that IP won't work.
> 
> So either comment out the line: `nano /etc/hosts` and change `104.199.65.124 ap-gew4.spotify.com` into `#104.199.65.124 ap-gew4.spotify.com` OR fix it via the code where it got set. `nano /usr/lib/systemd/system/spotify.service` and remove the line `ExecStartPre=/opt/hifiberry/bin/set-host-ip ap-gew4.spotify.com 104.199.65.124` then remove the current wrong resolved host by `nano /etc/hosts` and remove line `104.199.65.124 ap-gew4.spotify.com` then reboot.
> 
> Repo wise, th is solved here: #577 but it will be for the 64 version I guess.

This worked for me. Now I can use `Spotify Connect` again. I have a Raspberry Pi 3 and have updated to the new HifiBerryOS-64 Alpha. Spotify apparently no longer works with older versions of HifiBerryOS.

Unfortunately, the `Spotify Daemon` doesn't work, so I can't establish a local connection.

hyperbart, 2024-09-15:
Thank you @diederikdq .
I came here after 2 hours of Googling and fumbling with Spotify connect not working, thinking it was an issue on my part because I had a switch of ISP last week.

Removed the set IP from the service unit file and from the hosts file, everything works again.

And now I am also aware a new version of HifiberryOS is coming up and this explains why the version I was running didn't get any updates anymore. What a night :) .

jakubstrama, 2024-09-19:
> Hi This is discussed here https://support.hifiberry.com/hc/en-us/community/posts/21348779832605-Spotify-Connect-doesn-t-work-anymore-HifiberryOS-20230404?page=1. Chances are your issue will most [...]

This indeed solved this issue. Thanks iederikdq.

nextflex, 2024-12-02:
I tried Spotify Connect (premium account) it with hifiberryos 20210531 and alpha8 on Raspberry4. In both versions I can see the hifiberry in my Spotify App. When I start streaming on the device it hanging in a connecting state. In both version it helped to commend out the static IP /etc/hosts.

Weidav, 2025-02-05:
This is fixed in the Development branch, why is there no new release?
Everyone is running into this annoying bug and for people without "coding experience" it is quite a hurdle to fix.

diederikdq, 2025-06-30:
> I have an Pi 3 which still doesn't work after removing the IP. I thought this was due to the older version, so i borrowed also a Pi 4 and installed the new Hifiberry OS vanilla. Tried Spotify Connect: still the same issue "connecting". Removed the IP in the Pi 4 in the hosts file: still the same issue. What now?
> 
> [@diederikdq](https://github.com/diederikdq) [@Weidav](https://github.com/Weidav) [@jakubstrama](https://github.com/jakubstrama) do you still have this issue?

@hyperbart 
Which of the 2 paths did you follow?
Because it was either commenting out the line OR to adapt code and remove the IP. If you remove the IP, it was added back due to other code putting it back. If it is in comment, then it stays in comment.

I cannot answer if there are still issues:
Hifiberryos had a fork of https://github.com/ashthespy/librespot (which was a fork of librespot) but because that fork wasn't updated in the last 6y, I am guessing that's why the builtin one is having issues. I did use the spotifyd extension for a while but there I had other issues.
So with the https://www.hifiberry.com/blog/hifiberryos-quo-vadis/ a few months back, I temporarily switched to Moode where they use librespot and those issues do not happen. My reasoning: if hifiberry is doing a whole overhaul, updating spotify repo's will (understandably) not be top prio.