hifiberry/hifiberry-os #578: Update Support Doc on HiFiBerry Site

user10072023github, 2024-09-14:
**Describe the bug**
Support doc here: https://www.hifiberry.com/docs/software/how-to-establish-a-ssh-connection-with-a-raspberry-pi/
should include default SSH login details for HiFiBerryOS of username: root/password: hifiberry

**HiFiBerryOS version**
N/A

**HiFiBerry sound card**
N/A

**To Reproduce**
N/A

**Expected behavior**
Easy method of finding SSH login information in order to run diagnostics etc

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
Info to update is in the grid at the bottom of the page

hifiberry, 2024-09-14:
This has nothing to do with HiFiBerryOS. Password is set by the user in the UI.

closed 2024-09-14 by hifiberry

user10072023github, 2024-09-14:
Future readers can judge how insanely incompetent this response was.  For those poor souls, I'll memorialize the info here...if you don't set a password in the OS (SSH enabled by default), and you need to SSH in, then the info is as stated above (username: root, password: hifiberry).  I haven't done it yet, but when you are logged in successfully, you'll apparently be in the wrong directory (probably /root/, so you'll have to type "cd /" without the quotes to get to where you need to be in order to execute any useful commands.  Again, haven't done it so if anyone else knows feel free to jump in