hifiberry/hifiberry-os #582: Amp4 Pro not supported in 32bit Version

philipp2310, 2024-10-03:
**Describe the bug**
Amp4 Pro not supported in 32bit Version of HifiberryOS, this is caused by the not updated Kernel. Without apt by design, a manual update is not possible. A Kernel after 04.2024 seems required.

**HiFiBerryOS version**
20230404

**HiFiBerry sound card**
Amp4 Pro

**To Reproduce**
Any 32bit installation with Amp4 Pro.
64bit works (but is missing DLNA which is crucial for me)

**Expected behavior**
Amp4 Pro should work for both versions, 32bit and 64bit.

**Screenshots**
-

**Browser (if applicable)**
-

**Additional context**
-

hifiberry, 2024-10-03:
We do not plan any further version of the 32bit HiFiBerryOS. Upgrading the kernel would require a great effort that we better sepdn on HBOS64.

closed 2024-10-03 by hifiberry

philipp2310, 2024-10-03:
I completely understand, thank you.

Maybe a short information on some site would be nice, as the only thing I found after many hours of analyzing, reading logs, switching sound settings and searching the web was one community post. Followed by trying to get a manual kernel upgrade started without any of the usual linux tools (sudo, apt..)

A few ideas where it could be added:
Your old "[Updating the linux kernel](https://www.hifiberry.com/docs/software/updating-the-linux-kernel/)" page is not mentioning HifiberryOS(e.g. by design not possible, please wait for an official release), 
the git is not mentioning any compability to the possible hardware(e.g. OS32 compatible with everything but Amp4pro, please use OS64),
the shop is not mentioning any software requirements at all (and in general is missing the "details" page you got for example for the amp100).

I was quite close to sending the hardware back, when there was not a single sound in 4 hours of trying. Not a threat or something, I'm just mentioning it, because the setup itself doesn't need a tech savy person and is so easy(Awesome job! Linux and sound is hell, but you made it easy!), but my blind search for a solution was quite the opposite. Glad it works in OS64 without any issues just by plug and play!