hifiberry/hifiberry-os #584: Update Vollibrespot version

diederikdq, 2024-10-13:
**Is your feature request related to a problem? Please describe.**
I have installed the 64 alpa8 version on my hifiberries. There are some errors reported frequently
`[Vollibrespot] : couldn't parse packet from HIFIBERRYIPv4:5353: type 47 is invalid`

So I thought to check for updates and noticed that https://github.com/hifiberry/hifiberry-os/blob/development/buildroot/package/vollibrespot/vollibrespot.mk still mentions:
`VOLLIBRESPOT_VERSION = 5d0aa600d4af3d808a51e56df2bc6d7ee953a494`
Should I assume then that the current vollibrespot on alpha8 still points to your own forked Vollibrespot's commit of Jan 25, 2022: https://github.com/hifiberry/Vollibrespot/commit/5d0aa600d4af3d808a51e56df2bc6d7ee953a494 ?

Your fork on Vollibrespot has received updates since then. Can we update this version to the lastest version `VOLLIBRESPOT_VERSION = 96d169e0eb1d5019a224456fcd85554a97e716ed` ? Or can't that be used for some reason?

**Describe the solution you'd like**
Have vollibrespot use its latest version if possible.

**Describe alternatives you've considered**
None

**Additional context**
None

Profoktor, 2024-12-25:
vollibrespot has not seen any a activity for the last two years. Switching to the spotifyd extension seems a more promising way to go

hifiberry, 2025-01-21:
HiFiBerryOS64 has both vollibrepot and spotifyd (via extension) integrated.

diederikdq, 2025-02-08:
I don't want to login to 1 specific account with spotifyd, which is why I prefer the vollibrespot/librespot approach.
It's true that vollibrespot hasn't been updated for a while. It is built on librespot though. And librespot is still very well maintained.

I can't pinpoint why vollibrespot was used instead of librespot. So would it make more sense to look into using that in the future?

But apart from that, you have a more recent version so if there is no reason to use the old one then why not update it?

Profoktor, 2025-02-08:
spotifyd provides just what you're looking for (you don't need to register with the account, you can connect from the Spotify App via Spotify Connect)

Besides, a major update of spotifyd is on the way, which will update it to use a more recent version of librespot. Looking forward to this, and hopefully a new update of Hifiberry as well :)

Regarding your last remark - Updating the version would mean you
- have to adapt it
- have to test it and make sure that it works at least "as good as before"
- and it would still use an old librespot version - so, no benefit from librespot being well maintained
I can understand if there is no one willing to invest the time for an unmaintained library, especially if there is a viable alternative at hand

P.S. - if you want to give "plain librespot" a try, have a look at moode distro. But let me tell you that lot's of convenience regarding spotify is missing (in the end, you are only able to control via a Spotify App on a Smartphone / PC). At least for me that was one argument to remain on Hifiberry ;)

diederikdq, 2025-02-09:
@Profoktor Thanks! You're right. I was able to use without logging in.

The sound always starts full blast and I can't use it as is but I will investigate spotifyd a bit and see how to achieve what I could with Vollibrespot.
There I could modify /etc/vollibrespot.conf and set parameters to have linear volume control and more importantly initial-volume to be 50. And those would persist.
Was a bit of a hack to avoid every time I connected to have the volume to be full blast and scaring all my neighbours.

Anyway. I will close this. I can't expect HifiBerryOS to take over Vollibrespot and bring it up to current code of librespot (and bringing up to date with their own repo latest version is just a bandaid that might not be worth it indeed).

closed 2025-02-09 by diederikdq