hifiberry/hifiberry-os #585: rpi5/2G not booting a8

morteh, 2024-10-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Pi crashes with "Asynchronous SError Interrupt" while probing broadcom (bcm2712_pull_config_set)

**HiFiBerryOS version**
64alpha8

**HiFiBerry sound card**
n/a

**To Reproduce**
plug in freshly written SD card
power on 

**Expected behavior**
Expected to boot

**Screenshots**
[image]

**Browser (if applicable)**
n/a

**Additional context**
Note, this is the 2GB Pi5, slightly newer than the 4/8G models (there's my punishment for being cheap).

There are multiple somewhat similar-sounding reports of crashes probing broadcom, see:
https://forums.raspberrypi.com/viewtopic.php?t=365809
https://github.com/raspberrypi/linux/issues/5455
https://lore.kernel.org/all/20230502231558.5zt5tyxczd22ppjz@mraw.org/#t

I haven't tried a patched kernel, I'll update when/if I get around to it.

Only keyboard, monitor and power connected, stock raspbian boots fine from same SD card

kevindlau, 2025-01-17:
@morteh did you manage to solve this issue? I'm seeing the same problem with a new Pi 5.

Gearhed, 2025-01-29:
same problem here :(

revkorf, 2025-03-21:
And same problem here as well with a 4GB Pi5 Model B Rev 1.1.