hifiberry/hifiberry-os #586: 64Alpha8: Bluetooth: Volume not consistent after disconnect-reconnect

baldrianbandit, 2024-10-29:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Bluetooth volume is relatively low in general, I think. When connecting and setting a volume level, then disconnecting and reconnecting, the volume level is low again and - according to pulsemixer on my bluetooth client - different (lower) every time I reconnect it. 

**HiFiBerryOS version**
64Alpha8

**To Reproduce**
1. Connect via bluetoothctl connect <ADDR>
2. Start pulsemixer on client, see volume level, close pulsemixer
3. Disconnect client, then reconnect
4. Start pulsemixer on client, see volume level

**Expected behavior**
Bluetooth should either remember its last volume level or start at a fixed level on client connect.

Another suggestion: Using spotify-connect a lot, I disabled volume control of vollibrespot, so its output responds to global volume control only. I'd prefer that for bluetooth as well.

HifiberryOS is great! Thank you!

hifiberry, 2024-12-02:
While this is a common problem as different BT devices use different volume implementations. Unfortunately there's nothing we can do about this to.

baldrianbandit, 2025-02-12:
@hifiberry Workaround: Revert volume when device connects via bluetooth

1) Create udev rule
```
$ /etc/udev/rules.d/bt-add-event.rules
ACTION=="add", SUBSYSTEM=="bluetooth", TAG+="systemd", ENV{SYSTEMD_WANTS}="bt-add-event.service"
```
2) Create systemd unit
```
[Unit]
Description=Runs when bluetooth connection is added
Requires=bluetooth.target
After=bluetooth.target

[Service]
ExecStart=/usr/bin/bt-add-event.sh
Type=exec
Restart=always

[Install]
WantedBy=multi-user.target
```

3) Create script to reset volume
```
#!/usr/bin/sh
amixer sset DSPVolume 250
```
Further idea: Save volume on disconnect to a file and restore on reconnect