hifiberry/hifiberry-os #587: rest api for media control from other applications

agent-seed, 2024-11-02:
**Is your feature request related to a problem? Please describe.**
I'm trying to build a management console for all the systems in my house, it'd be very nice to be able to pause/skip songs via an api instead of relying on the HiFiBerry WebUI

**Describe the solution you'd like**
A rest api, ideally json, with post parameters for common actions (volume, input, play state, skipping) and get parameters for song info (title/artist, album art, current input, current volume level, possibly time elapsed/remaining) 

**Describe alternatives you've considered**
I could create a bluetooth remote for my phone to handle playback management, but this wouldn't have the ability to display song info and wouldn't work across devices. Another option would be embedding the webui as an iframe, but this wouldn't be as seamless of a solution  

**Additional context**
optional api security would be a nice-to-have, but probably wouldn't be necessary for most users

blockbasti, 2024-11-06:
This already exists
https://github.com/hifiberry/audiocontrol2/blob/master/doc/api.md

agent-seed, 2024-11-07:
Oh! I completely missed that!

closed 2024-11-07 by agent-seed