hifiberry/hifiberry-os #592: [feature] Add switch to turn off video / HDMI ports to save power

marcaurele, 2025-01-15:
**Is your feature request related to a problem? Please describe.**

I would appreciate if there was an option to disable the video ports on the raspberry base os so that I can save power when used in a headless mode. This should be set at the boot time, and would not find it strange to require to reboot the unit to activate the port(s) again.

**Describe the solution you'd like**
I would like a configuration option in the UI interface to state that the device is used in headless mode, not attached to any screen. Therefore a boot config option should be set/changed to turn the video off and require a reboot to be effective.

**Describe alternatives you've considered**
I looked at the commands that need to be run to turn it off, but none of them are available on the base image.
I believe the `/boot/config.txt` can be manually customized to comment out or change the lines starting with `dtoverlay...` to some values, which I'm not yet sure about. I might try...

**Additional context**

- https://forums.raspberrypi.com/viewtopic.php?t=283717
- https://www.raspberrypi.com/documentation/computers/config_txt.html#hdmi-audio
- https://www.raspberrypi.com/documentation/computers/config_txt.html#dtoverlay
- https://www.raspberrypi.com/documentation/computers/configuration.html#part3.1

hifiberry, 2025-01-21:
I'm sorry, but this is such a niche request that we do not plan to implement this. If you want to develop the code including UI integration, we're happy to look into a pull request

marcaurele, 2025-01-21:
Could you give me pointers to the code / project where such feature should be added and I'll try my best to come with a PR.

hifiberry, 2025-01-22:
Start building the system and adding the require command line tools. Check out the developers section here:
https://github.com/hifiberry/hifiberry-os/tree/master/doc